dbrgn/RPLCD #20: Support for i2c

DieterKoblenz, 2015-04-06:


dbrgn, 2015-04-06:
Thanks for that eloquent feature request description ;)

The feature is underway (see the [i2c branch](https://github.com/dbrgn/RPLCD/tree/i2c)). I might find some time to continue this week, but no guarantees.

DieterKoblenz, 2015-04-06:
Hehe, sorry I typed a whole description but that seemes to have been lost in the interwebs somewhere!

I absolutely love this lcd library but I need to use more then available GPIO ports therefor this request.

dbrgn, 2015-04-07:
For the record, this question should be answered once an i2c branch is merged: http://raspberrypi.stackexchange.com/questions/28991/how-can-i-use-a-lcd-display-with-a-i2c-adapter

marksev1, 2016-01-09:
Here are some i2c lcd libraries from which you could port support:

https://github.com/CaptainStouf/raspberry_lcd4x20_I2C

https://github.com/alinleonard/LCD-I2C-Raspberry-Pi-2

dbrgn, 2016-01-10:
@marksev1 that's not the problem, I can access I2C devices through the smbus library. the problem is that I've lacked the time to finish the implementaition (there were still some bugs around) :)

The good news is that an i2c display is lying on my desk right now, and that I want that to work with RPLCD in the next few weeks. So expect an update in a few days or weeks (can't promise any timeline though).

marksev1, 2016-01-10:
Looking forward to it. Have an off-topic question, I'm "scraping html" to get text from our national weather forecast service, I'm currently printing to the terminal. But eventually I would like to display it on the 20x4 i2c lcd. Since the lcd has room for 80 characters and the forecast has around 160 how would I do it? By using scrolling?

Would it be possible to split the text in pages? That could be an idea too right.

Also the lcd only supports ASCII right, since we have some different characters in our alphabet - č š ž.

dbrgn, 2016-01-10:
> Since the lcd has room for 80 characters and the forecast has around 360 how would I do it? By using scrolling?

Either scrolling (see https://blog.dbrgn.ch/2014/4/20/scrolling-text-with-rplcd/) or paging. I'd probably use paging and switch the page every 5-10 seconds. You can use regular Python string slicing to implement the pages.

> Also the lcd only supports ASCII right, since we have some different characters in our alphabet - č š ž.

For special characters, see [writing special characters](https://github.com/dbrgn/RPLCD/#writing-special-characters) to be able to view the entire character ROM. If the character isn't available yet, you can [create custom characters](https://github.com/dbrgn/RPLCD/#custom-characters) yourself.

if you have more questions, please open a separate issue.

dbrgn, 2016-12-08:
I2C support has landed (testet with PCF8574 based I2C port expander)!

Usage example:

    from RPLCD.i2c import CharLCD
    lcd = CharLCD(address=0x3f, port=1)

The rest should be the same as in GPIO mode.

closed 2016-12-08 by dbrgn

dbrgn, 2016-12-08:
[image]

marksev1, 2016-12-09:
Oooo, thank you!

PR #34 (2016-12-08): I2C Support
Yay, finally!

Fixes #20.

The commonly used I2C boards have backlight control onboard, so no GPIO pins are necessary (refs #8, #21).

To do:

- Document (probably in a separate PR)
- Add a backwards compatible interface with deprecation warnings