dbrgn/RPLCD #24: Offset line 3 and 4

iUltimateLP, 2015-12-25:
Hey, first of all, your library is great, but I have a problem with it.
I already worked with this LCD with the Arduino, and I had the same problem there, the 3rd and 4th line is offset by 4 characters, so where a letter would be in the first col, it is in the 4th col. Back with the Arduino, I fixed it by changing the RAM addresses for every single line, whereas the original address in the code was 0xXX, and I needed to change it to 0xYY (sorry, don't remember the values), which can be looked up in the datasheet of the LCD. So I searched through your code, but I didn't find any variable to specify the RAM adress for line 3 and 4, so could you add such variables in the code? Would be very handy, thanks!

iUltimateLP, 2015-12-25:
Oh, I am actually that dumb and gave a wrong col count :joy: Silly me

closed 2015-12-25 by iUltimateLP

dbrgn, 2015-12-26:
Hehe, that's what I thought :) Good that you found the fix yourself.