dbrgn/RPLCD #29: LCD 16x2: garbage or spontaneous settings change

vitasam, 2016-09-14:
Hi,
sometimes (e.g. after desk lamp, connected adjacent outlet, is switched ON0) LCD spontaneously changes settings. E.g. cursor becomes visible. Sometimes there are corrupted symbols.
Is this a (i) RasPi power source problem, (ii) long wiring of breaboard or (iii) software problem?
[image]

dbrgn, 2016-09-14:
Hi. I'm not sure. It could be either an unstable power supply, or some current induced through the long wires.

I'd first try another power supply to see if the problems disappear.

vitasam, 2016-09-15:
Yes, I have to find another power supply. What is a best way to re initialize this LCD without deleting the object and calling LCD constructor again?

dbrgn, 2016-09-15:
Maybe `lcd.clear()`?

vitasam, 2016-09-15:
I made such function, called after each button press:
```
def updateDisplay(a_line1, a_line2, a_clear_flag):
    if a_clear_flag:
        lcd.clear()
    lcd.cursor_pos = (0, 0)
    lcd.write_string(a_line1)
    lcd.cursor_pos = (1, 0)
    lcd.write_string(a_line2)
```

but apparently it did not help with desk lamp.

dbrgn, 2017-05-13:
@vitasam did you find a solution in the meantime?

dbrgn, 2017-07-30:
Closing this due to lack of activity.

closed 2017-07-30 by dbrgn