dbrgn/RPLCD #32: Mention 16x2 tutorial for wiring up

thijstriemstra, 2016-10-11:
This looks like a good tutorial: https://learn.adafruit.com/drive-a-16x2-lcd-directly-with-a-raspberry-pi/overview

thijstriemstra, 2016-10-11:
I will adjust the pins table in the readme and add a column for the 16x2, having to look it up on external sites isn't ideal.

thijstriemstra, 2016-10-11:
@dbrgn

I'm using the following pins with this [device](http://www.ebay.com/itm/-/171983948287?roken=cUgayN&soutkn=Q6RQtH), a 1602 16x2 Character LCD Display Module HD44780 Controller Blue Arduino.

Does this look right to you? I've tried running your test but the backlight turns on when I start the raspberrypi and nothing changes during the test_16x2.py. Any idea?

```rst
+------------+--------------------+---------------------------+
| HD44780    | Pi GPIO            | Comments                  |
| Pin Number | Pin Number (BOARD) |                           |
+============+====================+===========================+
| 1          | 6                  | GND                       |
+------------+--------------------+---------------------------+
| 2          | 2                  | VCC - 5V not 3.3V         |
+------------+--------------------+---------------------------+
| 3          | Any free GPIO pin? | VO - Contrast adjustment  |
+------------+--------------------+---------------------------+
| 4          | 15                 | Register Select           |
+------------+--------------------+---------------------------+
| 5          | 18 or 9            | R/W. Can be directly      |
|            |                    | attached to GND as well.  |
+------------+--------------------+---------------------------+
| 6          | 16                 | Enable                    |
+------------+--------------------+---------------------------+
| 7-10       | *Not connected*    | In 4-bit mode, we don't   |
|            |                    | connect them.             |
+------------+--------------------+---------------------------+
| 11         | 21                 | D4                        |
+------------+--------------------+---------------------------+
| 12         | 22                 | D5                        |
+------------+--------------------+---------------------------+
| 13         | 23                 | D6                        |
+------------+--------------------+---------------------------+
| 14         | 24                 | D7                        |
+------------+--------------------+---------------------------+
| 15         | 5V                 | Backlight LED Anode (+)   |
+------------+--------------------+---------------------------+
| 16         | GND                | Backlight LED Cathode (-) |
+------------+--------------------+---------------------------+
```

dbrgn, 2016-10-12:
> This looks like a good tutorial

Thanks for the link. There are pros and cons:

- The old Arduino based tutorial shows everything on one page whereas the Pi tutorial spreads the content over multiple pages
- The Pi tutorial "requires" some strange Adafruit breakout adapter. That might confuse users. What do you think?

Regarding this issue:

> I will adjust the pins table in the readme and add a column for the 16x2, having to look it up on external sites isn't ideal.

I disagree for the following reason:

Not all modules have the same pinout. If you buy a Char LCD on Aliexpress and blindly trust that it has the same pinout, you might fry your LCD (and maybe also your Pi?). It is important to always check the datasheet provided by the LCD manufacturer.

The Adafruit Arduino wiring tutorial mentions this and also suggests a way to calculate a resistor value, if one is needed. Therefore I think it's better if we don't provide too much information, otherwise users will rely on it and some will be disappointed when it does not work or breaks things.

Opinions?

thijstriemstra, 2016-10-20:
Eventually I used this video tutorial: http://www.rototron.info/lcd-display-tutorial-for-raspberry-pi/ with his GPIO pins and it worked. Backlight can't be controlled though (or not sure how), it's always at brightest.

dbrgn, 2016-10-20:
You can probably control the backlight brightness by using a potentiometerin front of the supply voltage, or by using PWM. Apparently there's [software PWM](http://raspi.tv/2013/rpi-gpio-0-5-2a-now-has-software-pwm-how-to-use-it) support in the RPi.GPIO library. There should also be hardware PWM pins on the Raspberry Pi, which would be even better.

But that's probably something that should be discussed in #8.

thijstriemstra, 2016-10-26:
Here's a [Fritzing drawing](http://fritzing.org/projects/raspberry-pi-16x2-lcd-with-contrast-control) for a 16x2 lcd with contrast control.

[image]

dbrgn, 2016-10-26:
Thanks! I'm quite busy at the moment, but I'll try to take a look soon-ish.

thijstriemstra, 2016-10-26:
This [tutorial](http://www.raspberrypi-spy.co.uk/2012/08/16x2-lcd-module-control-with-backlight-switch/)  also adds backlight brightness control and on/off switch.

[image]

dbrgn, 2016-12-10:
@thijstriemstra the docs are now ready, if you want you can improve them with your illustrations! Just make sure not to document anything as given that might be different on some of the LCD modules available :)

Phreak91, 2018-12-08:
while displaying date and time lcd shows garbage value and the screen constantly starts flashing.....but i solved it by adding a delay of about 80ms and clearing the lcd after the lcd write command. This is just a temporary solution...figuring out a permanent one....
here's my code
while True:
lcd.write_string("Time: %s" %time.strftime("%H:%M:%S"))
time.sleep(0.08)
lcd.clear()

dbrgn, 2019-03-04:
@Phreak91 your issue was probably resolved in #91.

I'll close this issue for now, PRs for improved docs are welcome!

closed 2019-03-04 by dbrgn