dbrgn/RPLCD #36: MicroPython Port

dbrgn, 2016-12-08:
Would be great.

thijstriemstra, 2016-12-08:
why.. what is micropython?

dbrgn, 2016-12-08:
A Python interpreter that runs on microcontrollers: https://micropython.org/

cfaessler, 2017-01-24:
@dbrgn is there already any progress on this? I need to control a HD44780 with PCF8574 with my LoPy.
Im thinking about to start to working on this. However, I need something to have something working by next Monday :)

dbrgn, 2017-01-24:
No, but it shouldn't be too hard, since the I2C functionality is already in there.

You could copy `RPLCD/i2c.py` to `RPLCD/micropython.py` to get started. Common code *could* be moved into a `common_i2c` file, but I don't think there's actually enough code in the i2c module for this to be worthwhile.

cfaessler, 2017-01-24:
Ok, what are other critical parts?
- I2C: is very straightforward in MicroPython, not a lot to do there
- __future__: Just saw that there is a port https://github.com/micropython/micropython-lib/blob/master/__future__/__future__.py
- Context managers: https://github.com/micropython/micropython-lib/tree/master/contextlib

I guess I start working on this Thursday.

dbrgn, 2017-01-24:
The [context managers](https://github.com/dbrgn/RPLCD/blob/b19cfefc32cfdfeb1780603bb82f2c2f9117272f/RPLCD/contextmanagers.py) are deprecated anyways, so maybe you can avoid support for them in the first place?

I'm currently working on a [custom codec](https://github.com/dbrgn/RPLCD/tree/codec), but that's probably supported by MicroPython as well.

melbow, 2017-07-18:
Hi, any progress on this? I'd be willing to contribute too if the work can be split, just let me know

dbrgn, 2017-07-19:
@melbow Not yet. Feel free to give it a stab, I haven't found the time to do so yet in the last few weeks.

In case you have trouble with the enums, I think these could also be removed from RPLCD since they don't provide that much benefit but make porting harder.