dbrgn/RPLCD #41: Compatible LCDs

cfaessler, 2017-01-24:
What about adding a list of tested devices/controllers in the README? I know there are minor differences, although devices are HD44780 compatible. Commands are usually not a problem, but addresses (such as CGRAM) could be.

dbrgn, 2017-01-24:
The problem is that most LCDs don't really have a product name, right?

cfaessler, 2017-01-24:
No they don't. But the controllers are known.

dbrgn, 2017-01-24:
That's true. Feel free to start such a list, the best place for it would probably be a new page [in the docs](https://github.com/dbrgn/RPLCD/tree/master/docs), with a link from README.

brantje, 2017-12-07:
I think a nice list can be found [here](https://www.aliexpress.com/wholesale?catId=0&SearchText=i2c+lcd) (aliexpress).

dbrgn, 2017-12-07:
> I think a nice list can be found here (aliexpress).

Just because all those displays use I2C does not mean that the protocol is compatible. It usually is, but there's not really a simple way to find out before trying it.

brantje, 2017-12-07:
Good catch. Maybe make a wiki page with tested lcds?
I have some at home, but need to test them first with this lib.

dbrgn, 2017-12-07:
I also have a few at home, but the main issue is, how do you identify an LCD? They're all no-brand, so the only thing we can differentiate is the I²C port expander (which we already list [here](https://github.com/dbrgn/RPLCD#supported-ic-port-expanders). And that only works if the LCD uses I²C, not parallel mode.

dbrgn, 2019-03-04:
Closing this for now, but pull requests are welcome.

closed 2019-03-04 by dbrgn