dbrgn/RPLCD #42: BCM numbering mode

nineclicks, 2017-03-13:
When trying to use BCM numbering mode I get the following error

```
  File "/usr/local/lib/python2.7/dist-packages/RPLCD/lcd.py", line 211, in __init__
    GPIO.setmode(self.numbering_mode)
ValueError: A different mode has already been set!
```

Is this expected behavior or should I be able to use BCM numbering?

dbrgn, 2017-03-13:
This isn't expected. But maybe another script / program is running at the same time that also uses the GPIO?

If not, it's probably a bug.

dbrgn, 2017-05-13:
I'll close this. Feel free to comment if the problem persists.

closed 2017-05-13 by dbrgn

tyeth, 2020-09-06:
Just got this, running the same script two minutes after closing and making a change, think I'm not releasing it properly as I used to have it wrapped in a **with**