dbrgn/RPLCD #44: I2C ImportError

berkutta, 2017-05-03:
I'm not 100% sure if the issue is on my side or the Library. I get the following error on executing my python script:
```
$ python main.py 
Traceback (most recent call last):
  File "main.py", line 5, in <module>
    from RPLCD.i2c import CharLCD
ImportError: No module named 'RPLCD.i2c'
```
It's a fresh Raspbian installation with python-smbus package, activated i2c kernel module and your python library. My Python script is under the following repo: https://github.com/berkutta/rackbuddy

dbrgn, 2017-05-03:
i2c is not deployed to pypi yet. I'll try to do that today.

dbrgn, 2017-05-03:
(Until then you can use `pip install -e git+https://github.com/dbrgn/RPLCD@master#egg=RPLCD-dev`)

berkutta, 2017-05-03:
Works, thanks!

closed 2017-05-03 by berkutta

dbrgn, 2017-05-03:
I'm still working on #40, want to get that done for the 1.0.0 release :)

dbrgn, 2017-05-09:
@berkutta version 0.9.0 is finally on PyPI!