dbrgn/RPLCD #46: Install from git somehow broken

fake-name, 2017-05-09:
On a fresh install from git:

```

pi@testpi:~$ sudo pip3 install git+https://github.com/dbrgn/RPLCD.git
Collecting git+https://github.com/dbrgn/RPLCD.git
  Cloning https://github.com/dbrgn/RPLCD.git to /tmp/pip-vatij0r9-build
^NInstalling collected packages: RPLCD
  Running setup.py install for RPLCD ... done
Successfully installed RPLCD-0.4.0

Traceback (most recent call last):
  File "test_20x4.py", line 7, in <module>
    from RPLCD import gpio, i2c
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/__init__.py", line 5, in <module>
    from .gpio import CharLCD as GpioCharLCD
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/gpio.py", line 30, in <module>
    from .lcd import BaseCharLCD
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/lcd.py", line 27, in <module>
    from . import codecs
ImportError: cannot import name 'codecs'

```

Somehow, the `codecs` directory isn't getting created. I suspect that the setuptools dependency resolution isn't detecting the codecs files.

fake-name, 2017-05-09:
I think the breaking commit is https://github.com/dbrgn/RPLCD/commit/d4dc083363986c3d9d754f44aabf9c86471bea9a

dbrgn, 2017-05-09:
Ah, yeah, thanks for this issue! I think the problem is a missing MANIFEST.in entry or something like that.

I'll make a new pypi release hopefully today. If not today, then definitely tomorrow.

fake-name, 2017-05-09:
Coo.

For the moment, I just got lazy and vendored the whole library. It works fine from the git folder.

closed 2017-05-09 by dbrgn

dbrgn, 2017-05-09:
Release 0.9.0 is out on PyPI! A simple `pip install RPLCD` should work now.

dbrgn, 2017-05-09:
> For the moment, I just got lazy and vendored the whole library. It works fine from the git folder.

Sure, that's fine. If you want, you can now check out the `v0.9.0` tag to get the stable version.

> Having this makes my life a lot easier, thanks for releasing it!

Great to hear :)

> As a side-note, the 20x4 test python file is broken:

Oh, thanks for letting me know. I'll fix :)

fake-name, 2017-05-09:
Wait, did you go from 0.4.0 to 0.9.0? What happened to 0.5.0 - 0.8.0?

dbrgn, 2017-05-09:
Yep, I plan to release 1.0.0 soon, and 0.9.0 is something like the "release candidate".

fake-name, 2017-05-09:
Heh, I see. I kind of guessed it was something like that.

It sure seems a hell of a lot more stable then most raspberry pi stuff I've run across.

dbrgn, 2017-05-09:
> It sure seems a hell of a lot more stable then most raspberry pi stuff I've run across.

Thanks! That was one of my main goals. Most scripts for the Pi copy-pasted from some tutorials and consist of very hacky code. I wanted something clean and pythonic, with an easy API.

fake-name, 2017-05-10:
Just installed from github, it seems to work fine.

Only complaint I have at this point is there's no `.__version__`, `.version` or similar, so it's hard to tell what version I'm actually using. That really just a nicety, though.

dbrgn, 2017-05-11:
You can check `setup.py` or `` :) If it's important to you, feel free to open an issue for it.

fake-name, 2017-05-11:
The issue I had here was more "Ok, I /think/ it installed over my manual version, how do I verify what version I'm running". 

Basically, I had a *installed* instance that seemed to work, but I wasn't sure if it was actually the one from pypi or one I installed manually. Once the package is installed, there doesn't seem to be an easy way to determine the version, short of finding the module location and looking in the source file/at the egg manually.

PR #47 (2017-05-09): Fix setup.py, prepare for release
Fixes #46