dbrgn/RPLCD #60: Defaults for gpio

GoranLundberg, 2017-05-16:
The default pins for gpio should be removed.
Its really dangerous. I have accidentially run code several times with default pins. And with sensors, motordrivers etc connected to the pi. Im just lucky that nothing broke.

The user should explicitly have to set these manually.

These defaults seem obsolete from the beginning ofta the project with only one configuration.

Also. Gpio mode GPIO.BOARD as defaults should be removed or changed to GPIO.BCM that is way more used. Cant actually remember to have seen a project with GPIO.BOARD as default.

dbrgn, 2017-05-16:
> The default pins for gpio should be removed.

Agreed, you make a good point.

> Also. Gpio mode GPIO.BOARD as defaults should be removed or changed to GPIO.BCM that is way more used.

I find BOARD much more intuitive, but we could remove the default and make it a required parameter. (Changing the default to BCM would break all existing GPIO based code, I don't think that's a good idea.)

It would make the RPLCD API a bit more complex though.

GoranLundberg, 2017-05-17:
> I find BOARD much more intuitive, but we could remove the default and make it a required parameter. (Changing the default to BCM would break all existing GPIO based code, I don't think that's a good idea.)

Agree

> It would make the RPLCD API a bit more complex though.

Yeah, but not that much, only once when initiating the class.
If it keeps users from breaking their hardware I'm all for it :)

dbrgn, 2017-07-30:
This has been fixed in the meantime.

closed 2017-07-30 by dbrgn