dbrgn/RPLCD #61: Error when changing to GPIO.BCM mode using the example

4QStudios, 2017-05-18:
hi, every time i want to initialize the lcd in BCM mode for the pins it fails, is this hardcoded to work only in BOARD mode?
the file is  gpio.py

GoranLundberg, 2017-05-18:
I got it working in BCM mode. What does your code look like?

4QStudios, 2017-05-18:
this is the output when i set the BCM mode
[image]

Here you can see the code
[image]

GoranLundberg, 2017-05-18:
You have the wrong data pins. pins_data needs to be the BCM numbering on the Pi. For example 0 does not exist. Check this site for BCM numbering. https://pinout.xyz/

4QStudios, 2017-05-18:
im a retard, i was using the wPi number scheme instead of the BCM, im closing this

closed 2017-05-18 by 4QStudios