dbrgn/RPLCD #62: Unexpected keyword argument "numbering_mode"

mfnalex, 2017-05-28:
When using the following code, I get this error message:

> Traceback (most recent call last):
  File "./test.py", line 9, in <module>
    charmap='A02',auto_linebreaks=True)
TypeError: __init__() got an unexpected keyword argument 'numbering_mode'

code:
```
from RPLCD.i2c import CharLCD
import RPi.GPIO as GPIO

lcd = CharLCD(  pin_rs=12, pin_rw=6, pin_e=17, pins_data=[18, 22, 23, 24],
                numbering_mode=GPIO.BCM,
                cols=16,rows=2,dotsize=8,
                charmap='A02',auto_linebreaks=True)
```

dbrgn, 2017-05-29:
You probably want to use the GPIO based `CharLCD` class instead:

```python
from RPLCD.gpio import CharLCD
```

Then that code should work. In I²C mode there are no pin definitions and therefore also no numbering mode.

mfnalex, 2017-05-29:
Thanks, I will try it again. But if that is true, the Get-Starded page in the docs should be changed to use RPLCD.gpio instead of RPLCD.i2c for the GPIO part.

dbrgn, 2017-05-29:
> But if that is true, the Get-Starded page in the docs should be changed to use RPLCD.gpio instead of RPLCD.i2c for the GPIO part

oh, dang, I didn't notice that. Thanks for letting me know :)

> EDIT: I got rid of the error, but I cant get the display to show text, only a line of white boxes... :/

Did you set up a potentiometer to control the contrast? You should see the text once you turn it down a bit.

commit da389f9d (2017-05-29): Docs: Fix "getting started" instructions for GPIO

Fixes #62
closed 2017-05-29 by dbrgn