dbrgn/RPLCD #64: No module (After installing) error messages.

FrozenBeard, 2017-07-17:
I installed Via PIP as required and I already have python-smbus installed. 
I seem to have python2 and python3 on my Pi, I didnt have pip so I had to install it using sudo easy_install3 pip.

Installing this:
```
tom@raspberrypi:~/python $ sudo pip install RPLCD
Collecting RPLCD
  Downloading RPLCD-0.9.0-py2.py3-none-any.whl
Installing collected packages: RPLCD
Successfully installed RPLCD-0.9.0
```
I then wget the test_20x4.py and ran it under both python2 and python3. Neither work.

```
tom@raspberrypi:~/python $ sudo python test_20x4.py
Traceback (most recent call last):
  File "test_20x4.py", line 28, in <module>
    from RPLCD import gpio, i2c
ImportError: No module named RPLCD
tom@raspberrypi:~/python $ sudo python3 test_20x4.py
Traceback (most recent call last):
  File "test_20x4.py", line 28, in <module>
    from RPLCD import gpio, i2c
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/__init__.py", line 5, in <module>
    from .gpio import CharLCD as GpioCharLCD
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/gpio.py", line 27, in <module>
    import RPi.GPIO as GPIO
ImportError: No module named 'RPi'
```

dbrgn, 2017-07-18:
What distribution are you using on your Pi? On Raspbian, the RPi library should be preinstalled.

Can you try this?

    sudo apt install python-rpi.gpio python3-rpi.gpio

FrozenBeard, 2017-07-18:
Weird, Im using
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```
I have got Pi-hole installed if that makes any difference.

That install gave me:
```
The following NEW packages will be installed:
  python3-rpi.gpio
```
Ill install this and try again.

FrozenBeard, 2017-07-18:
Nope. Same deal.

```
tom@raspberrypi:~ $ sudo apt install python-rpi.gpio python3-rpi.gpio
[sudo] password for tom:
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-rpi.gpio is already the newest version.
The following NEW packages will be installed:
  python3-rpi.gpio
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 28.2 kB of archives.
After this operation, 182 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian/ jessie/main python3-rpi.gpio armhf 0.6.3~jessie-1 [28.2 kB]
Fetched 28.2 kB in 2s (11.1 kB/s)
Selecting previously unselected package python3-rpi.gpio.
(Reading database ... 33816 files and directories currently installed.)
Preparing to unpack .../python3-rpi.gpio_0.6.3~jessie-1_armhf.deb ...
Unpacking python3-rpi.gpio (0.6.3~jessie-1) ...
Setting up python3-rpi.gpio (0.6.3~jessie-1) ...
tom@raspberrypi:~ $ ls
python
tom@raspberrypi:~ $ cd ./python/
tom@raspberrypi:~/python $ ls
driver  examples.py  lcd_i2c.py  RPi_I2C_driver.py  RPLCDtest.py  test_20x4.py
tom@raspberrypi:~/python $ sudo python test_20x4.py
Traceback (most recent call last):
  File "test_20x4.py", line 28, in <module>
    from RPLCD import gpio, i2c
ImportError: No module named RPLCD
tom@raspberrypi:~/python $ sudo python3 test_20x4.py
Traceback (most recent call last):
  File "test_20x4.py", line 28, in <module>
    from RPLCD import gpio, i2c
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/i2c.py", line 25, in <module>
    from smbus import SMBus
ImportError: No module named 'smbus'
tom@raspberrypi:~/python $
```

dbrgn, 2017-07-18:
Yeah, it looks like RPi.GPIO is installed for Python 2 by default, but not for Python 3 :)

I'll close this issue, but feel free to comment if something does not work.

closed 2017-07-18 by dbrgn

gyscos, 2017-10-31:
Shouldn't this be part of the dependencies for this library? Relying on it being installed independently sounds a bit dangerous?

dbrgn, 2017-10-31:
The dependencies are conditional. The RPi.GPIO module is only required when accessing hardware via GPIO, and the smbus module is only required when accessing hardware via I²C.

This could probably be solved using pip categories(?), e.g. "pip install RPLCD[i2c]", I don't have any experience with those though. @gyscos maybe you?