dbrgn/RPLCD #65: test_20x4.py i2c

FrozenBeard, 2017-07-24:
Im using a pi1, I have to use i2cdetect 0 to get my i2c address of 0x27. In your docs you mention i2cdetect 1, so im assuming there is a variable to tell the framework that im using a Pi 1 and to look in the right place.

I have no idea if im using MCP23008 or PCF8574, but BOTH return the same errors in python.

```
tom@raspberrypi:~/python $ sudo python3 test_20x4.py i2c MCP23008 0x27 A02
Traceback (most recent call last):
  File "test_20x4.py", line 56, in <module>
    lcd = i2c.CharLCD(sys.argv[2], int(sys.argv[3], 16), cols=20, rows=4, charmap=sys.argv[4])
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/i2c.py", line 79, in __init__
    auto_linebreaks=auto_linebreaks)
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/lcd.py", line 94, in __init__
    self._init_connection()
  File "/usr/local/lib/python3.4/dist-packages/RPLCD/i2c.py", line 85, in _init_connection
    self.bus = SMBus(self.port)
FileNotFoundError: [Errno 2] No such file or directory
```

dbrgn, 2017-07-25:
Yes, that number is the `port` argument to `CharLCD`. Can you try adding `port=0` to [this row](https://github.com/dbrgn/RPLCD/blob/cec68b7856b7ff7c7d6b0fd7beefc7f15a7358c1/test_20x4.py#L56)?

dbrgn, 2017-07-30:
@FrozenBeard the testsuite has now been completely revamped by @GoranLundberg. The new test script should support setting the `port` parameter. Let me know if that doesn't work!

closed 2017-07-30 by dbrgn

FrozenBeard, 2017-12-07:
So I finally managed to get the test to fully work, after stumbling over every single commad in your docs written for python2 instead of python3, this finally worked for me.
`sudo python3 lcdtest.py i2c testsuite expander=PCF8574 port=0 cols=20 rows=4 addr=0x27`
Now I can write my own damn scripts!

dbrgn, 2017-12-07:
Actually if you install RPLCD for Python 2, then all the commands should work just fine with the `python` binary. But it's good to hear that you got it running!