dbrgn/RPLCD #66: Online docs are out of date

ozarchie, 2017-07-29:
The link to the online readthedocs points to a page that is out of date.
http://rplcd.readthedocs.io/en/stable/api.html
This doc shows a getting started example and API reference that are not consistent with the library.
This one pointing to a pdf is correct:
https://media.readthedocs.org/pdf/rplcd/latest/rplcd.pdf

dbrgn, 2017-07-30:
Yeah, that's because I planned to release the newest version of the library a while ago and somehow didn't find time for it.

I'll try to do it tonight, then it should be in sync again :)

ozarchie, 2017-07-30:
Thanks Danilo, great library.  Thanks so much. Cheers John

dbrgn, 2017-07-30:
1.0.0 is released, links to documentation are updated :)

> Thanks Danilo, great library.  Thanks so much. Cheers John

Thank you, feedback from users is very appreciated :)

closed 2017-07-30 by dbrgn