dbrgn/RPLCD #69: 40x2 LCD?

sphh, 2017-08-07:
Thank you for your gread LCD library!

Is it possible to use a 40x2 LCD? My guess is, it's not possible, since these LCDs need two enable lines and ``CharLCD.__init__`` only has one ``pin_e`` keyword …

Do you plan to add support for these LCDs? Thanks.

dbrgn, 2017-08-10:
That should not be a problem. I have one here, but haven't tested it yet.

Why does it need two enable lines? A 20x4 LCD is also just an 40x2 LCD internally, just with line wraps.

sphh, 2017-08-14:
Sorry, you are absolutely right: A 40x2 LCD has just one enable line!

I wanted to write "40x4", which should have two enable lines. There is one LCD library for the Arduino which adds support for these large displays: https://github.com/Swap-File/tron-suit/tree/master/Helmet/Software/Libraries/LiquidCrystalFast

Sorry for the confusion!

dbrgn, 2017-08-14:
I see. I don't currently have such a display and it's not currently supported. It *could* work maybe if you just create two CharLCD instances configured to the two separate enable pins and make sure that you don't write to both sections at the same time.

If someone wants to implement support for this, pull requests are welcome :) But I'll close this for now.

closed 2017-08-14 by dbrgn