dbrgn/RPLCD #70: Getting Garbage Display

vbisbest, 2017-09-29:
WHen doing several updates in a row, the output on the LCD becomes corrupted.  Garbage characters and the only way out is a full reset.    This is the relevant code:

`        lcd.clear()
        lcd.cursor_pos = (0, lineStartPos(lcdLine1))
        lcd.write_string(str(lcdLine1))
        lcd.cursor_pos = (1, lineStartPos(lcdLine2))
        lcd.write_string(str(lcdLine2))
        lcd.cursor_pos = (2, lineStartPos(lcdLine3))
        lcd.write_string(str(lcdLine3))
        lcd.cursor_pos = (3, 0)
        lcd.write_string("Target:")
        lcd.write_string(str(targetTemp))
        lcd.write_string(chr(223) + " ")
        lcd.write_string("Temp:")
        lcd.write_string(str(currentTemp))
        lcd.write_string(chr(223))`

lineStartPos is a function to determine center point of the text.   I can update the same content perhaps 6 times in a row but after that it becomes corrupt.   Thanks.

ageir, 2017-09-30:
Are you using GPIO directly or an I2C backpack?

wa4557, 2017-11-21:
@ageir Having the same problem. I'm using

```
lcd = CharLCD(pin_e=38, pin_rs=40, pin_rw=None,
              pins_data=[37, 35, 33, 31], numbering_mode=GPIO.BOARD,
              cols=16, rows=2)
```

Is that ok, or should I use something else?

thijstriemstra, 2018-04-15:
I'm also seeing this on 20x4 with GPIO using v1.1.0..

dbrgn, 2018-04-16:
@thijstriemstra is this reproducable?

thijstriemstra, 2018-04-16:
Yes, see my previous comment.

spellbin, 2018-04-17:
I also am having the same issue. First several displays work then it becomes garbled. I'll try reverting back to an older version and see if that helps.

thijstriemstra, 2018-04-17:
Yeah, I spoke too soon, reverting to 1.0.0 didn't fix it. It's an issue that does not happen all the time..

aagontuk, 2018-04-24:
I have encountered the same problem. Tried with different versions from 1.1.0 to 0.9.0. Got same results. For version 0.9.0 LCD backlight didn't work. Here is my code:

```
from RPLCD.gpio import CharLCD
from RPi import GPIO
from time import sleep

lcd = CharLCD(pin_rs = 15, pin_e = 16, \
                pins_data = [21, 22, 23, 24], \
                numbering_mode = GPIO.BOARD, \
                cols = 20, rows = 4, dotsize = 8, \
                charmap = 'A00', auto_linebreaks = True, \
                pin_backlight = 18, \
                backlight_mode = 'active_high')

try:
    while True:
        lcd.clear()
        lcd.write_string('hello, world...')
        sleep(5)
except KeyboardInterrupt:
    lcd.close(clear = True)
```

dbrgn, 2018-04-24:
Ok, good to know, that means it's probably not a new issue. I'll try to have a look once I find some free time for it :)

@aagontuk @thijstriemstra @spellbin do you by chance use very long cables from the LCD to the Raspi?

aagontuk, 2018-04-24:
I am using short cables. May be this is not a h/w or wiring issue. [Adafruit CharLCD](https://github.com/adafruit/Adafruit_Python_CharLCD) is working fine without any issue yet.

dbrgn, 2018-04-24:
Adafruit's library uses [busy looping as timing function](https://github.com/adafruit/Adafruit_Python_CharLCD/blob/master/Adafruit_CharLCD/Adafruit_CharLCD.py#L305-L309) while RPLCD uses [time.sleep](https://github.com/dbrgn/RPLCD/blob/master/RPLCD/common.py#L98-L105). Maybe that's the problem.

spellbin, 2018-04-24:
I'm using short cables as well.

albedozero, 2018-12-20:
I know this issue is a bit stale, but FYI I've been having the same problem and tried forking and replacing time.sleep in the [msleep and usleep functions](https://github.com/dbrgn/RPLCD/blob/7cb5693605a55ac018e16d6512fe83c831febc0e/RPLCD/common.py#L98) with a [busy loop as in the Adafruit CharLCD library](https://github.com/adafruit/Adafruit_Python_CharLCD/blob/master/Adafruit_CharLCD/Adafruit_CharLCD.py#L305-L309). The garbage display problem still crops up for me, so maybe that's not it.

lcheng1, 2019-02-12:
Could anybody give my PR a try to see if it solves their issues? I've had it running on a RPi 3 with my fix for about two weeks now and have yet to see any scrambled characters.

dbrgn, 2019-02-13:
The pull request #91 was merged to master. It adds a compatibility mode with longer wait times. This might help when an LCD requires timings that are out of spec for the HD44780, or if the clock isn't correct.

It would be great if one of the affected people could test the current master branch of this library, to see if it fixes the issue. You also need to pass in the `compat_mode=True` parameter when initializing the `CharLCD` class in order to enable this compatibility mode.

albedozero, 2019-02-13:
The master branch fixes the problem for me, when I use `compat_mode=True`. Works in python2 and python3.

dbrgn, 2019-02-14:
Great, in that case I'll make a release today!

Thanks @lcheng1 for tracking this down.

closed 2019-02-14 by dbrgn

dbrgn, 2019-02-14:
v1.2.1 is out.

DolceVii, 2019-06-06:
I still have that issue on v.1.2.2 
compat_mode=True ....nothing change

phil1283, 2020-03-15:
is there a way to use compat_mode with IIC ???
Thanks

val43, 2020-11-28:
> Are you using GPIO directly or an I2C backpack?

I'm having this issue with a 16x2 display with a I2C backpack.
Setting a 0.1s delay between commands seem to get rid of the problem.

PR #91 (2019-02-13): Added a legacy mode.
Older LCDs (or LCDs that are based off the HD44870) aren't all made equal.

It appears that some don't run at the reference clock, and as such, they're in a busy state far more often than newer or better ones, which means that, occasionally, subsequent sends are missed. 

I based this off of Adafruit's older library, which included a flat 1 millisecond delay on every send event. From my testing, about 0.5 milliseconds is enough to prevent the issue, but I erred on the higher side at 0.7 milliseconds here. 

From my testing, this fixes #70 which seems to plague only some us [... cut]