dbrgn/RPLCD #73: Can't seem to use? Keeps saying no module?

TommyR-GB, 2018-01-13:
Apologies as I am a total noob, but I am trying to use this and I keep getting an error message when running this code:

from RPLCD.gpio import CharLCD
lcd = CharLCD(cols=16, rows=2, pin_rs=37, pin_e=35, pins_data=[40, 38, 36, 32, 33, 31, 29, 23])
lcd.write_string(u'Hello!')

This is my error message, and I have followed the PIP installation process...

Traceback (most recent call last):
  File "/home/pi/Desktop/Pi Projects/LCD Control Basics.py", line 1, in <module>
    from RPLCD.gpio import CharLCD
ImportError: No module named 'RPLCD'

Can you tell me what I am doing wrong?

TommyR-GB, 2018-01-13:
Closing this - I'm stupid. Trying to run it on Python 3, having installed it on Python 2.7, have now installed on Python 3 by using the command sudo pip3 install RPLCD (just in case anyone else is daft enough to encounter this issue)

Still cant get it to work - but I'm a good step closer...

closed 2018-01-13 by TommyR-GB

dbrgn, 2018-01-14:
@TommyR-GB good that you could figure it out. if you have any question, you can also ask in [our Gitter channel](https://gitter.im/RPLCD/Lobby) :)