dbrgn/RPLCD #74: Does not work on Python 3.5.3

9291Sam, 2018-01-19:
Launching up the code and it works beautifully on python 2.7.13 but everything I run on 3.5.3 has the same error message even when I only put in the provided setup code. (Yes my lcd i2c address was 0x27)   
I run this code

from RPLCD.i2c import CharLCD
lcd = CharLCD(i2c_expander='PCF8574', address=0x27, port=1,
              cols=20, rows=4, dotsize=8,
              charmap='A02',
              auto_linebreaks=True,
              backlight_enabled=True)

and it always gives the error

Traceback (most recent call last):
  File "/home/pi/Screen.py", line 6, in <module>
    backlight_enabled=True)
  File "/home/pi/.local/lib/python3.5/site-packages/RPLCD/i2c.py", line 164, in __init__
    auto_linebreaks=auto_linebreaks)
  File "/home/pi/.local/lib/python3.5/site-packages/RPLCD/lcd.py", line 99, in __init__
    self.command(0x03)
  File "/home/pi/.local/lib/python3.5/site-packages/RPLCD/lcd.py", line 377, in command
    self._send_instruction(value)
  File "/home/pi/.local/lib/python3.5/site-packages/RPLCD/i2c.py", line 237, in _send_instruction
    (value & 0xF0)) | self._backlight)
OSError: [Errno 121] Remote I/O error
>>> 

on python 3.5.3 but it works perfectly fine on python 2.7.13
Not really sure what is going on here. I am running a Pi 3 Model B V1.2

dbrgn, 2018-01-19:
Hm, that's strange. Just out of curiosity, are you using Raspbian with standard kernel, or something else?

9291Sam, 2018-01-19:
Not 100% Sure I always used rpi-update here is uname -a 
pi@raspberrypi:~ $ sudo uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

dbrgn, 2018-01-19:
I'm not sure if this is related, but I recently noticed that the SHT21 temperature sensor kernel driver worked on the stock kernel, but not on the latest (installed with `rpi-update`).

After some searching I found out that the driver uses I²C clock stretching, and this:

- http://www.advamation.com/knowhow/raspberrypi/rpi-i2c-bug.html
- http://www.bitwizard.nl/wiki/Bcm2835_i2c_bug

Maybe Python 3.5.3 uses I²C in a different way than Python 2, so that this bug is triggered?

Do you have a spare SD card around that you could use to test whether the problem exists on stock Raspbian as well?

9291Sam, 2018-01-19:
I can do that I will get stock raspbian from raspberry pi site and try it

9291Sam, 2018-01-19:
Also could it matter that I am using the NOOBS boot loader ( it just came with the pi)

dbrgn, 2018-01-19:
No, NOOBS is just a bootstrapping tool to install a Linux distribution, it should install the same Raspbian as if you would download the image directly :)

9291Sam, 2018-01-20:
I Just decided to reinstall the library fro 3.5.3 and it didn't work
I then reinstalled the library on 2.7.13 and rebooted and i works
my guess is that the .whl file it installs from using pip was corrupted

9291Sam, 2018-01-20:
as of about 2 minutes ago it broke in the same way again so...

9291Sam, 2018-01-30:
After some more debugging I finally narrowed it down to the code giving an error because there was no lcd connected to the i2c port and it appeared a wire had come loose

closed 2018-01-30 by 9291Sam