dbrgn/RPLCD #76: Scroll text vertically

cubetastic33, 2018-02-05:
Is there any way I can scroll text vertically on my 20x4 lcd? Like the first line goes away and the next line appears down, and so on?

dbrgn, 2018-02-05:
Sure, but nothing built-in. You can use a line buffer (similar to this: https://blog.dbrgn.ch/2014/4/20/scrolling-text-with-rplcd/) and then shift the lines in the line buffer up or down. And finally, write the resulting buffer entirely to the LCD.

Does that help?

cubetastic33, 2018-02-05:
@dbrgn Ah! I already read that post, but didn't think it would help. But ok, I'll try it out!

cubetastic33, 2018-02-05:
@dbrgn do you think you could give me a function that could do that? 😃

dbrgn, 2018-02-05:
Nope, sorry 😉 You'll have to do the work yourself.

But here's the general idea:

- You have a list of 4 strings, every string is 20 characters long
- You write all those 4 strings to the LCD
- Then you modify the list, for example with the `pop()` and `append()` methods (see https://docs.python.org/2/tutorial/datastructures.html#more-on-lists)
- Then you write those 4 strings to the LCD again
- ...and so on

cubetastic33, 2018-02-06:
@dbrgn thanks a lot! I managed to do it!

dbrgn, 2018-02-06:
Great to hear! 🙂

closed 2018-02-06 by dbrgn