dbrgn/RPLCD #84: CursorMode Not Working

vickey97, 2018-08-09:
Hey, I was interfacing my LCD with RPi and used this library but when i run the code below it gives me the error. 'ImportError: cannot import name 'CursorMode'. If this is changed to something else how can i get it working?

`from RPLCD import CharLCD
from RPLCD import CursorMode
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BOARD)
GPIO.setup(33,GPIO.OUT)
GPIO.setup(31,GPIO.OUT)
GPIO.setup(29,GPIO.OUT)
GPIO.setup(23,GPIO.OUT)

lcd=CharLCD(numbering_mode=GPIO.BOARD, pin_rs=37, pin_e=35, rows=2, cols=16, pins_data=[33,31,29,23])
try:
    lcd.clear()
    lcd.cursor_pos = (0, 0)
    lcd.write_string(u'This Is My First\n\rLCD Interfacing!')
    lcd.cursor_mode = CursorMode.blink
except:
    pass

finally:
    GPIO.cleanup()
`
Any help is greatly appreciated. Thank You!

dbrgn, 2018-08-09:
Hi @vickey97!

Your code only worked with an older version of RPLCD. Here you can find the current documentation on setting the cursor mode:

http://rplcd.readthedocs.io/en/stable/usage.html#changing-the-cursor-appearance

You can simply delete the second line (`from RPLCD import CursorMode`) and change the line `lcd.cursor_mode = CursorMode.blink` to `lcd.cursor_mode = 'blink'`.

Does that help?

vickey97, 2018-08-09:
Yeah This Helps! Thanks a lot.

closed 2018-08-09 by vickey97

dbrgn, 2018-08-09:
By the way, here's a complete "getting started" guide:

http://rplcd.readthedocs.io/en/stable/getting_started.html
http://rplcd.readthedocs.io/en/stable/usage.html