dbrgn/RPLCD #86: Can D0-D3 be used for data?

vickey97, 2018-09-15:
Hey, I actually read every tutorial there is and all of them uses D4-D7 pins on 16x2 LCD for 4 bit method. I just wanted to ask that will my program run if i use D0-D3 pins instead of D4-D7 for 4 bit data?

Any help is greatly appreciated.
Thank you!

dbrgn, 2018-09-15:
As far as I know, the controller will only use D4-D7 in 4-bit mode, and using D0-D3 will simply not work :) Check the datasheet of the HD44780 for details.

closed 2018-09-15 by dbrgn

SlySven, 2020-01-02:
You could use the LCD in 8-bit mode which will use all of D0-D7 and is a bit faster - but the physical interface to the LCD module is ***designed*** to only use the high nibble when it is set to 4-bit mode. 

Indeed setting the interface back from 8-bits to 4-bits needs care if you only have the 4-bit interface to work with.

One thing to bear in mind is that when used with, say, a PCF8574 I2C backpack those only have 8 pins and two (or three or four if you need to control a backlight and/or be able to read from as well as write to to the device) will be needed to work the interface alongside the 4 data lines...