dbrgn/RPLCD #89: smbus2 support

joscha, 2018-11-26:
Any chance a pull request that allows usage of `smbus2` as a drop-in replacement would be accepted? Potentially also as a fallback when `smbus` can not be loaded (or as a favored in case `smbus2` is installed). I am trying to make RPLCD work on Python 3 with balena and I am having a hard time getting the smbus packages in the right state.

dbrgn, 2018-11-26:
Hi. What are the advantages of `smbus2` over `smbus`?

dbrgn, 2018-11-26:
Ah, that is cool :slightly_smiling_face:

I could imagine something like this working well:

```python
try:
    import smbus
except ImportError:
    import smbus2 as smbus
```

What do you think?

joscha, 2018-11-26:
Yep, that would be great!

dbrgn, 2018-11-26:
Would you be willing to create a pull request?

Simply updating all the imports would be sufficient :slightly_smiling_face: Additionally, updating the README / Docs to say "smbus or smbus2" would be nice too.

joscha, 2018-11-27:
> Simply updating all the imports would be sufficient 🙂 Additionally,

Of course, will do this evening (I am UTC+10).

joscha, 2018-11-27:
PR is in https://github.com/dbrgn/RPLCD/pull/90 - let me know if there is any other place I should mention smbus2

PR #90 (2018-11-27): feat: support smbus2 as drop-in replacement for I2C
closes #89
commit 7cd8850b (2018-11-27): Support smbus2 as drop-in replacement for I2C access (#90)

Fixes #89.
closed 2018-11-27 by dbrgn