dbrgn/RPLCD #93: Cursor position (0, 20) invalid on a 4x20 LCD

dbrgn, 2019-03-03:
When initializing an LCD with "auto_linebreaks=False", then writing a 20 character string to position `(0, 0)` seems to fail.

```python
Traceback (most recent call last):
  File "/usr/local/bin/lcd.py", line 60, in <module>
    main(controller, addr, cols, rows)
  File "/usr/local/bin/lcd.py", line 28, in main
    loop(lcd, unit)
  File "/usr/local/bin/lcd.py", line 48, in loop
    lcd.write_string(datestring.ljust(lcd.lcd.cols))
  File "/usr/local/lib/python3.5/dist-packages/RPLCD/lcd.py", line 279, in write_string
    self.write(char)
  File "/usr/local/lib/python3.5/dist-packages/RPLCD/lcd.py", line 406, in write
    self.cursor_pos = newpos
  File "/usr/local/lib/python3.5/dist-packages/RPLCD/lcd.py", line 151, in _set_cursor_pos
    raise ValueError(msg.format(pos=value, lcd=self.lcd))
ValueError: Cursor position (0, 20) invalid on a 4x20 LCD.
```

dhrone, 2019-03-24:
You need to decide what you are going to do when auto_linebreaks is true and you have completely filled a line.  I'd suggest returning col to 0.  Perhaps change line 404 in lcd.py to...
`                newpos = (row, col + 1 if col + 1 < self.lcd.cols else 0)`

dbrgn, 2019-03-24:
I'm not sure. My preferred solution would be to simply continue writing to the memory. If you turn off auto linebreaks, then you're on your own and any logic should be disabled.

This means that we'd have to disable bounds checks in that case.

joscha, 2019-05-15:
I agree with @dbrgn, once auto linebreaks are off I think if you write more cols than there are it shouldn't be the problem of the library any more.

dbrgn, 2019-05-15:
Pull requests are welcome! I don't currently have much time for working on this bugfix.

JsBergbau, 2020-04-05:
Anything new when this issue will be fixed in the official release you install with pip? Even resetting the Cursorposition doesn't help. Second write leads to an exception. Since I return the Cursorposition to the beginning, this shouldn't happen. If you wriite "ab" in the second write, display shows "ab3456789abcdefghijk", so moving the cursor seems to work.
```
lcd.write_string("123456789abcdefghijk")
lcd.home()
lcd.cursor_pos=(0,0)
lcd.write_string("123456789abcdefghijk") #Exception
```

val43, 2020-11-28:
This is also applies to my 16x2 display
is there any workaround, other than just not using the right-most char?

dbrgn, 2020-11-28:
> is there any workaround, other than just not using the right-most char?

Maybe using `auto_linebreaks=False`.

> Anything new when this issue will be fixed in the official release you install with pip?

Pull requests fixing this issue would be welcome! See [this comment](https://github.com/dbrgn/RPLCD/issues/93#issuecomment-475963819) on how it should be fixed.

dbrgn, 2021-10-03:
Oops, it seems that in my previous comment I overlooked that there's already a proposed fix in #109. Let's continue the discussion there.

poggenpower, 2022-09-26:
Hi,

have the same issue and took a look into the code. Shouldn't this code 
https://github.com/dbrgn/RPLCD/blob/master/RPLCD/lcd.py#L404-L420
ensure that the cursor is not running out of its scope with `auto_linebreaks=True`?
If I understand the code correct will the cursor move to the next line if the current is filled and if the last line is filled it flips to POS (0, 0).
Not sure why it is failing.

dbrgn, 2023-03-25:
Should be fixed with #109.

closed 2023-03-25 by dbrgn