dbrgn/RPLCD #96: Japanese Characters Only when using Qapass 1602a LCD on Raspberry Pi 3B and Python 3.5

thecortex, 2019-03-25:
Hello,

I have tried using the charmap A00, and A02.

I have tried connecting in 4-bit, and 8-bit modes.

The problem is that all English characters are displayed like Japanese.

The numbers and punctuations are displayed absolutely right.

I am using Raspberry Pi 3B and connected directly to the Qapass 1602a LCD.

Hope you have any hints why I am getting this issue...

kroesi, 2019-03-25:
Hello thecortex,
all the character sets of HD44780 and clones I know do have ASCII numbers for letters... Are you sure you didn't swap some data pins and chose the right numbering mode (board vs. bcm)? This could definitly lead to your problem.

dbrgn, 2019-03-26:
I assume you're using GPIO (parallel) mode. Can you run the `rplcd-tests gpio show_charmap <options>` tool to show the charmap and then create and upload a photo of the LCD?

thecortex, 2019-03-26:
@kroesi, thank you for the hint, I'm using bcm mode, following the attached chart.

[image]

```
from RPLCD import CharLCD
from RPi import GPIO
from time import time

GPIO.setwarnings(False)
lcd = CharLCD(numbering_mode=GPIO.BCM, cols=16, rows=2, pin_rs=26, pin_e=19, pins_data=[18, 23, 24, 25, 12, 16, 20, 21], charmap='A00')
#lcd.write_string(u'{}'.format(time()))
lcd.write_string(u'1234567890123458')
lcd.cursor_pos = (1, 0)
lcd.write_string("abcdefghijklmnopq")
```

@dbrgn, the tests are available only if I clone the repository from GitHub, but what if I use the Pip package. I will give it a shot anyways, thanks for the hint...

thecortex, 2019-03-26:
Here is my wiring just in case something looks obviously wrong [image]

thecortex, 2019-03-26:
@dbrgn , thanks for the hint, it seems the characters are all displayed correctly, so we have the right wiring,and let's focus on the software side.

I have run the tests using a git clone of the master branch of the repository, and here were the results:

```
$ ./rplcd-tests gpio show_charmap cols=16 rows=2 mode=BCM rs=26 e=19 data=18,23,24,25,12,16,20,21
This tool shows the character map of your LCD on the display.
Press ctrl+c at any time to abort.

Displaying page 1 (characters 0-31). Press <ENTER> to continue.
Displaying page 2 (characters 32-63). Press <ENTER> to continue.
Displaying page 3 (characters 64-95). Press <ENTER> to continue.
Displaying page 4 (characters 96-127). Press <ENTER> to continue.
Displaying page 5 (characters 128-159). Press <ENTER> to continue.
Displaying page 6 (characters 160-191). Press <ENTER> to continue.
Displaying page 7 (characters 192-223). Press <ENTER> to continue.
Displaying page 8 (characters 224-255). Press <ENTER> to continue.
Test done. If you have a programmable backlight, it should now be off.

```

[image]

dbrgn, 2019-03-26:
@thecortex if you look at the [HD44780 datasheeŧ](https://cdn-shop.adafruit.com/datasheets/HD44780.pdf) (pages 17 and 18) you can see the standardized A00 and A02 charmaps. From quick comparison it looks like it's A00, but in wrong order, right?

The reason for that could be mixed-up address wires (D0-D7). Can you double-check them? Make sure that the data pins are specified the correct order (`data=D0,D1,...,D7`).

If I can see that correctly, the green wire goes from D7 to GPIO20, but in your data pin list 20 is not the last entry.

thecortex, 2019-03-27:
@dbrgn thank you very much it works now very well

closed 2019-03-27 by thecortex