dbrgn/RPLCD #100: print hex FF not working

riker65, 2019-04-13:
Hi
I want to print hex FF from charset A00: whole block
running: `lcd.write_string('\xFF')`

I expected the whole block to be written?

please check

or what am I doing wrong?

Thx

dbrgn, 2019-04-14:
With that you will only print the character at position 0xff of the A00 charset. To print the whole block you have to add every character to the string.

To show the entire character map on your LCD, you can use the `show_charmap` target of the `rplcd-tests` script that should get installed together with the RPLCD library.

dbrgn, 2019-04-14:
I'll close this, feel free to leave a comment if things are still unclear.

riker65, 2019-04-14:
Hi Danilo

maybe I was unclear.
I just want to print one sign. post FF in Charset A00.

But when I do
`lcd.write_string('\xFF')`
the sign - a single full lcd - is not printed.

How to print the hex FF?

Thanks Thomas

dbrgn, 2019-04-15:
Ah, I see. Are you sure that your LCD uses the A00 charset? What character is printed instead?

Are you using Python 2 or 3?

riker65, 2019-04-15:
Hi Danilo,

come code:
`lcdmode = 'i2c'
cols = 20
rows = 4
charmap = 'A00'
i2c_expander = 'PCF8574'
address = 0x27 # If you don't know what yours is, do i2cdetect -y 1
port = 1 # 0 on an older Pi`

`i = 0
while i < 20:
    lcd.cursor_pos = (0, 0)
    lcd.write_string('bsxFF:')
    lcd.cursor_pos = (3, i)
    sleep(0.1)
    i = i +1
    lcd.write_string('\xFF')

sleep(2)`

nothing is printed.

thanks for checking

riker65, 2019-04-15:
running python 2

dbrgn, 2019-04-15:
Can you try ` lcd.write_string(u'\xFF')` instead?

riker65, 2019-04-15:
Hi,
also nothing printed

dbrgn, 2019-04-15:
But does printing regular text work? Have you tried using the `rplcd-tests` script that's installed with RPLCD?

riker65, 2019-04-15:
di not try that 
but regular text and custom chars are working fine

riker65, 2019-04-15:
rplcd-tests i2c show_charmap col=20 row=4
not running