dbrgn/RPLCD #101: pigpio mode doesnt work on Python 2

magenois, 2019-04-25:
getting this error when running in pigpio mode on Python 2.7

```
Traceback (most recent call last):
  File "lcdtest.py", line 8, in <module>
    lcd = CharLCD(pi, pin_rs=26, pin_e=19, pins_data=[12,16,20,21])
  File "/usr/local/lib/python2.7/dist-packages/RPLCD/pigpio.py", line 154, in __init__
    auto_linebreaks=auto_linebreaks)
  File "/usr/local/lib/python2.7/dist-packages/RPLCD/lcd.py", line 94, in __init__
    self._init_connection()
  File "/usr/local/lib/python2.7/dist-packages/RPLCD/pigpio.py", line 226, in _init_connection
    self._writescript = self.pi.store_script(bytes(piscript, 'utf-8'))
TypeError: str() takes at most 1 argument (2 given)
```

Changing line 226 in pigpio.py from:
`self._writescript = self.pi.store_script(bytes(piscript, 'utf-8'))
`
to :
`self._writescript = self.pi.store_script(piscript.encode('utf-8'))`

fixes the issue.

bytes() function is not backward compatible to Python 2

dbrgn, 2019-04-26:
Pull requests to fix this are welcome!

Note that Python 2 will be officially unsupported in less than a year (January 1, 2020): https://python3statement.org/

dbrgn, 2020-04-03:
Since Python 2 is now officially out of support, Python 2 fixes won't be added anymore.

closed 2020-04-03 by dbrgn