dbrgn/RPLCD #102: Displays works sometimes and sometimes is blank or garbled.

seanspotatobusiness, 2019-05-20:
I am using a Raspberry Pi to send text to a 16x2 display which uses a SPLC780D controller. Sometimes the text displays correctly and sometimes the display is blank or garbled. Someone on the Raspberry Pi forum suggested that the timings assumed by the RPLCD library might not be correct. Is it possible to tweak these? Below is a video showing the problem and below that is the Python script I'm using.

I'm using RPLCD 1.2.2 on Raspbian Stretch.

https://www.youtube.com/watch?v=uTN3Vad8qDI

`import RPi.GPIO as GPIO
from RPLCD.gpio import CharLCD
import time

while True:
    lcd = CharLCD(pin_rs=36, pin_e=38, pins_data=[31, 33, 35, 37],
              numbering_mode=GPIO.BOARD,
              cols=16, rows=2, dotsize=8,
              auto_linebreaks=False,)
    lcd.write_string("abcdefghijklmnop\r\nqrstuvwxyz012345")
    time.sleep(1)
    lcd.clear()`

dbrgn, 2019-05-20:
Hi. Can you try setting `compat_mode=True` in the `CharLCD` constructor? See #70 for more details.

seanspotatobusiness, 2019-05-20:
Yes, this fixes the problem. Thanks very much.

dbrgn, 2019-05-20:
Great. We should improve the docs, I opened #103 for that.

closed 2019-05-20 by dbrgn