dbrgn/RPLCD #106: BPI LCD 1602 display module

PedroCavaleiro, 2019-08-22:
I've just acquired the [BPI LCD 1602 display module](http://wiki.banana-pi.org/BPI_LCD_1602_display_module) but I can't get it to work correctly

I have an Raspberry PI B rev 2

For that I used the following code

```
from RPLCD.i2c import CharLCD

lcd = CharLCD(i2c_expander='MCP23017', expander_params={'gpio_bank': 'B'}, address=0x20, port=1, cols=16, rows=2, charmap='A02')
lcd.clear()
lcd.home()
lcd.cursor_mode = 'blink'
lcd.write_string('Hello World')
```

Is there any hints how to get this module to work?

With this code I get the following result
[image]

dbrgn, 2019-08-23:
Hm, I'm not sure. Are you sure the expander_params and the address are correct?

PedroCavaleiro, 2019-08-23:
Get partially working, the cursor now moves correctly but when I use write_string I can see that the cursor moves to the correct position but still no text

dbrgn, 2019-08-23:
Maybe the data is written to the wrong memory location. But I don't know the details of this display, sorry.

PedroCavaleiro, 2019-08-23:
How can I overwrite the memory location, to try a new location?

JBrummans, 2021-02-12:
@PedroCavaleiro did you happen to solve this issue?
I have a very similar LCD using the MCP23017 but with [a red PCB and "Designed for RPI"](https://www.seeedstudio.com/Raspberry-Pi-RGB1602-LCD-Module.html)

I get the same result as you with the blinking cursor.

PedroCavaleiro, 2021-02-12:
> @PedroCavaleiro did you happen to solve this issue?
> I have a very similar LCD using the MCP23017 but with [a red PCB and "Designed for RPI"](https://www.seeedstudio.com/Raspberry-Pi-RGB1602-LCD-Module.html)
> 
> I get the same result as you with the blinking cursor.

Yes, l managed to get the LCD and buttons working, I will send the code as soon as I can

dbrgn, 2021-02-12:
@PedroCavaleiro good to hear that you could resolve your issue! I assume it was not a problem with RPLCD?

I'll close this issue for now. Feel free to leave a comment how you solved it, or whether it was a problem related to RPLCD.

closed 2021-02-12 by dbrgn

jyap808, 2025-03-12:
I have a similar device. Here is my working Go code:

https://github.com/jyap808/charLCDRGBI2C/