dbrgn/RPLCD #107: PCF8574t  it will work??

9acca9, 2019-09-09:
Hi, i found in you library that
"Supported port expanders are the PCF8574, the MCP23008 and the MCP23017."

Sorry my ignorance, but it will work with a "PCF8574t"??? I have to change something? if yes, what??

(i dont speak english)
Thanks!

dbrgn, 2019-09-09:
I'm not sure, you'd have to check out the datasheets to compare the functionality, or simply try it out.

I would guess that it should work, but I don't know what the difference between versions with the `t` and versions without are.

Let me know if you need anything else.

closed 2019-09-09 by dbrgn

SlySven, 2020-01-02:
Of course the OP could be referring to the PCF8574**A** which is functionally identical to the one without the suffix but *which has a different range of possible I2C addresses* - but in this case it is because he has an SMD (Surface Mount Device) rather than a DIL (Dual In-Line - i.e. with pins that go into holes on a printed circuit board) one - as we can see from the [documentation](https://www.nxp.com/docs/en/data-sheet/PCF8574_PCF8574A.pdf):

[image]

for the addresses used the details are:
[image]
[image]