dbrgn/RPLCD #110: Missing Characters

raysp, 2020-04-03:
I am having the same problem as mentioned by others with missing characters on the display.

Using GPIO without  compat_mode=True  exposes the problem.

using GPIO with  compat_mode = True solves the problem.

However I would like to use pigpio and with it I seem to have the same issue (perhaps
worse).  

Any ideas.

dbrgn, 2020-04-03:
Hi @raysp, can you test whether the branch at #98 solves your problem?

raysp, 2020-04-03:
I would love to try this.  I'm new to all of this package installing.  ( Before retirement I was
"in management" so someone else did this.)

I have been using pip3 to install.  Could you give me the command needed.

Thanks

dbrgn, 2020-04-03:
@raysp sure! Pip can install directly from git:

    pip3 install git+https://github.com/makslevental/RPLCD@compat_mode_pigpio

raysp, 2020-04-03:
now not working at all:

pi@raspberrypi:~ $ python lcd.py
Traceback (most recent call last):
  File "lcd.py", line 11, in <module>
    ,charmap='A02'              )
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
161, in __init__
    compat_mode=compat_mode)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/lcd.py", line
110, in __init__
    self.command(0x03)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/lcd.py", line
401, in command
    self._send_instruction(value)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
340, in _send_instruction
    self._send(value, c.RS_INSTRUCTION)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
311, in _send
    if self.compat_mode:
AttributeError: 'CharLCD' object has no attribute 'compat_mode'

pi@raspberrypi:~ $ more lcd.py
import pigpio
from RPLCD.pigpio import CharLCD
pi = pigpio.pi()

import time
pi = pigpio.pi()

lcd = CharLCD(pi,pin_rs=23,pin_e=24 \
,pins_data=[12,16,20,21]    \
,cols=16,rows=2             \
,charmap='A02'              )

lcd.cursor_mode = 'blink'
count = 0
lcd.write_string('xx8765')
while count < 100:
    print (count)
    lcd.clear()
    cf = 'POS:{}'.format(count)
    print (cf)
    lcd.write_string(cf)
    time.sleep(1)
    count = count + 1

lcd.write_string('Done')
pi.stop()

dbrgn, 2020-04-03:
Ah, I think if you had RPLCD already installed it won't get overwritten with the version from the branch.

Can you try this?

    pip3 install -U git+https://github.com/makslevental/RPLCD@compat_mode_pigpio

(-U will upgrade the package)

raysp, 2020-04-04:
same error;
 pip3 install -U git+
https://github.com/makslevental/RPLCD@compat_mode_pigpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/makslevental/RPLCD@compat_mode_pigpio
  Cloning https://github.com/makslevental/RPLCD (to revision
compat_mode_pigpio) to /tmp/pip-req-build-tmqwaqi3
Branch 'compat_mode_pigpio' set up to track remote branch
'compat_mode_pigpio' from 'origin'.
Switched to a new branch 'compat_mode_pigpio'
Building wheels for collected packages: RPLCD
  Running setup.py bdist_wheel for RPLCD ... done
  Stored in directory:
/tmp/pip-ephem-wheel-cache-r9871ldc/wheels/e4/18/31/7208bb9bac0e8907f99e21aa461bceab9d725012a3b254c981
Successfully built RPLCD
Installing collected packages: RPLCD
  Found existing installation: RPLCD 1.2.2
    Uninstalling RPLCD-1.2.2:
      Successfully uninstalled RPLCD-1.2.2
Successfully installed RPLCD-1.2.2

pi@raspberrypi:~ $ python lcd.py
Traceback (most recent call last):
  File "lcd.py", line 11, in <module>
    ,charmap='A02'              )
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
161, in __init__
    compat_mode=compat_mode)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/lcd.py", line
110, in __init__
    self.command(0x03)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/lcd.py", line
401, in command
    self._send_instruction(value)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
340, in _send_instruction
    self._send(value, c.RS_INSTRUCTION)
  File "/home/pi/.local/lib/python3.7/site-packages/RPLCD/pigpio.py", line
311, in _send
    if self.compat_mode:
AttributeError: 'CharLCD' object has no attribute 'compat_mode'
pi@raspberrypi:~ $ pip3 install -U git+
https://github.com/makslevental/RPLCD@compat_mode_pigpiopip3 install -U git+
https://github.com/makslevental/RPLCD@compat_mode_pigpio

raysp, 2020-04-20:
Just a note on this issue.   Apparently the default install location does
not have the latest version.

I installed it from git an the issue of requiring a contrast pin went
away.  This missing characters
still appear when this is run on a PI 4.  I just got a Zero W and (perhaps
not surprisingly)  pigpiod
works fine on this model.