dbrgn/RPLCD #111: Unreleased fix for contrast

TojikCZ, 2020-04-28:
Hi, I need to include your lib in a project, but want the contrast controll turned off. I see the code is fixed but not released. Like this, I can either:
- Include the git path in my requirements making my project use potentially unstable code
- Fork this and release it myself freezing that version in time
- Specify the contrast pin, not use it and mark it as used artificially lowering the available pins

Or I can ask you to make a release. Could you please please please do that?
Thank you

dbrgn, 2020-04-29:
Hi. Yep, you're right, I'll definitely try to make a release soon!

I'm quite busy today and tomorrow, so if I forget about this, please remind me by the weekend.

TojikCZ, 2020-04-29:
Will do, thank you for the quick reply

TojikCZ, 2020-04-30:
Hi, I'm facing skipped characters, but the pigpio and gpio implementations are different, one not supporting the compat thing. Moreover, there is a freaking while loop without a sleep in there. That's brutal, in my intended application, the pi has to have a lot more CPU left over. I'm gonna fork this and come back when I have something "better". Maybe wait with that release after all. :D

dbrgn, 2020-04-30:
@TojikCZ if you can get a PR up for that while loop by the weekend, I can probably include that in the release, yes 🙂 and otherwise I can simply create a second release next week.

Unfortunately I don't use the pigpio backend myself, so I fully rely on other testers / contributors.

dbrgn, 2020-05-03:
Version 1.3.0 is released.

closed 2020-05-03 by dbrgn