dbrgn/RPLCD #117: Character map does not seem to match A02 tables

etokheim, 2020-07-19:
Please tell me if I've misunderstood, but I think there is a miss-match between the A02 tables and which characters are shown on the display.

When I try to write the Danish/Norwegian letters: æ/ø/å, the screen displays different characters. Check out the attached image, where line one corresponds to the characters on line two. (ae = æ, oe = ø and aa = å).

[image]

I wrote to the display using this command:
```lcd.write_string("ae, oe, aa\n\ræ   ø   å")```

_I used this [pdf](https://www.sparkfun.com/datasheets/LCD/HD44780.pdf), page 18, to reference which characters the A02 table contains._

dbrgn, 2021-03-10:
Sorry for the late response. It's possible that your display does not match the A00 or A02 tables, but contains a custom mapping.

As described here https://rplcd.readthedocs.io/en/stable/usage.html#character-maps you can use the `show_charmap` target of the `rplcd-tests` script to write all characters sequentially to the display. Then you can compare with the character maps in the PDF.

If there's a mismatch, you probably need to create your own character map, or alternatively use custom characters or raw character references.

Does that help, @janis-ax @etokheim?

etokheim, 2021-03-14:
Hey, @dbrgn. Thanks for the reply!

I haven't worked on this project since January, but I will definitely try your suggestions when I resume development. I guess it's likely that my display has some sort of custom mapping...

I can, and have created some custom characters as a temporary fix, but as far as I know you can only create up to 8 custom characters. That's unfortunately not enough, though I've fixed the most common letters.

dbrgn, 2021-03-14:
@etokheim @janis-ax alright, I'll close this issue for now! You can still leave comments though.

closed 2021-03-14 by dbrgn

astroganga, 2022-11-26:
Hi, I have the same issue on a 20x04 i2c display. Only the A00 table match with documentation. A02 does not match. Sadly I cannot use latin letters of my language.

dbrgn, 2022-11-26:
Hi @astroganga, [this reply](https://github.com/dbrgn/RPLCD/issues/117#issuecomment-795298887) is still valid. 

Note that you can also [create up to 8 custom characters](https://rplcd.readthedocs.io/en/stable/usage.html#creating-custom-characters) if you're missing some character.