dbrgn/RPLCD #118: auto_linebreaks is broken

semmelsamu, 2020-08-04:
Code to reproduce:

```
from RPLCD.i2c import CharLCD

lcd = CharLCD(i2c_expander='PCF8574', address=0x27, cols=20, rows=4, auto_linebreaks=False)

lcd.write_string('HellorHellorHellorHelloHello')
```

Altough line breaks are disabled (`auto_linebreaks=False`), it breaks in the third row.

[image]

dbrgn, 2020-08-05:
It's not broken, that's how the layout of the LCD works. Internally, a 4x20 LCD is a 2x40 LCD, where the first row is split up into rows 1 and 3, and the second row is split up into rows 2 and 4.

closed 2020-08-05 by dbrgn

semmelsamu, 2020-08-05:
Still, `auto_linebreaks` should prevent that from happening. What else should it effect?

dbrgn, 2020-08-05:
No, with `auto_linebreaks` off the data is written directly to the controller memory. In that case you are responsible for setting the proper coordinates before writing data, depending on your LCD layout.

I think the behavior you want is precisely `auto_linebreaks=True`. In that case, the wrapping happens as expected.

dbrgn, 2020-08-05:
https://rplcd.readthedocs.io/en/stable/usage.html#automatic-line-breaks