dbrgn/RPLCD #119: Writing not possible, just background light control

rodentooth, 2020-08-29:
I use the backpack PCF8574**t** and a 20x4 blue white display. 
I have all required prerequisites, the i2c detects the device and the contrast is set to high. 
Yet, my display shows this on init, clear and any other command:

https://i.stack.imgur.com/YL57y.jpg

Backlight control works

dbrgn, 2020-08-29:
Hm, my guess would be that the contrast is set too high?

Do you control it using a pot on the backpack?

rodentooth, 2020-08-30:
Hi Daniel

Many thanks for your reply. I tried twiddling around with the contrast, not helping either. only the two full bars like in the image.
The contrast is set on the backback, yes

dbrgn, 2020-08-30:
Ok, then I don't know unfortunately what causes this, sorry. You might want to check whether there are significant differences between the PCF8574 and the PCF8574t.

closed 2020-08-30 by dbrgn