dbrgn/RPLCD #122: sudo and permissions

clach04, 2021-01-26:
Most distros have an i2c user group that can be assigned to users removing the need for sudo. https://lexruee.ch/setting-i2c-permissions-for-non-root-users.html has a good guide for machines that do not have this setup already.

Is this worth covering in the documentation? If yes, I can knock something up and raise a PR.

clach04, 2021-01-26:
BTW thanks for this project! There are a number of Python LCD modules, some do not handle i2c, and the others ONLY handle i2c. I love that this handles both (and it works on my hardware which is not a Raspberry Pi).

dbrgn, 2021-01-28:
> Is this worth covering in the documentation? If yes, I can knock something up and raise a PR.

As long as it's an approach that works on Raspbian (which 99% of the users will probably use), I think it would be a nice addition to the docs!

> BTW thanks for this project! There are a number of Python LCD modules, some do not handle i2c, and the others ONLY handle i2c. I love that this handles both (and it works on my hardware which is not a Raspberry Pi).

Great to hear!

clach04, 2021-01-30:
> > Is this worth covering in the documentation? If yes, I can knock something up and raise a PR.
> 
> As long as it's an approach that works on Raspbian (which 99% of the users will probably use), I think it would be a nice addition to the docs!

Yep, just checked. Raspbian creates permissions/groups out of the box :-)

    [pi@raspberrypi:~] $ cat /etc/os-release
    PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
    NAME="Raspbian GNU/Linux"
    VERSION_ID="9"
    VERSION="9 (stretch)"
    VERSION_CODENAME=stretch
    ID=raspbian
    ID_LIKE=debian
    HOME_URL="http://www.raspbian.org/"
    SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
    BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
    [pi@raspberrypi:~] $ grep i2c /etc/group
    i2c:x:998:pi

dbrgn, 2021-01-30:
Fixed in #124.

closed 2021-01-30 by dbrgn