dbrgn/RPLCD #127: clear(), home() and backlight off->on all fill the screen with white pixels

alan-rpi, 2021-08-07:
First, many thanks for making this LCD function available :)

I have installed RPLCD-1.3.0 using sudo pip3 install RPLCD. I have a 2004A V1.1 display with a PCF8754 +2037 and I'm using I2C on a Rasp Pi Zero w/o wifi under Buster 5.10.17+ #1421 

I can write text OK:

[image]

but when I use clear() or home() the screen is filled with white pixels and the text is very hard to read.

[image]

Turning the backlight off and then on also fills the screen with white pixels.

But in all cases, as soon as I write text it is OK again but **not** if the text is spaces or a null string. 

Could this be a bug or a faulty screen or me not understanding how to use RPLCD?

Best wishes,
Alan

dbrgn, 2021-08-16:
Hi. I haven't run across this behavior so far, so I don't really know what could cause this. The contrast is set up properly, right? Maybe it's simply an LCD controller knockoff that is not fully compatible with the original one by Hitachi.

dbrgn, 2021-08-16:
Maybe the "space" character in the character ROM is filled with the rectangle instead of the space. You could dump the entire character map on your LCD with the `show_charmap` target of the `rplcd-tests` script (which should be installed through pip, if I'm not mistaken).

alliefitter, 2021-10-05:
I ran into this same problem, though it was just `close()` that was causing the screen to white out. My problem ended up being that I had RW wired to ground because I'd just wired it up to an Arduino based on a tutorial, and that how it was wired. Hooking up RW and defining the pin in the `CharLCD` init fixed the problem. Just a little more detail, the screen did not immediately go all white as if I was messing with the contrast. In my case, once `clear()` was called, all characters (starting at 0, 0) would become entirely white sequentially at a pace of around one character every 300 to 500 milliseconds (guessing at the timing), and there was a pause of around a second after the first row was fully white. I'm a complete novice, so I don't know if wiring RW to GND is an improper usage of this Hitachi standard, but unplugging it from my Arduino and plugging it into my Pi and not defining the `pin_rw` resulted in this problem.

alan-rpi, 2021-10-05:
Thanks @dbrgn . Just seen your email - I didn't get an email notification. I have read there are problems with clones.
Thanks @alliefitter . Interesting that you had the same problem. I did not look to see if the screen went white sequentially. My screen came with the driver electronics (it uses an I2C interface to the pi). Like you I am a novice, so unsure if I can make the changes you made when I'm using I2C.

For now I have got round the problem by using ` lcd.clear()` and `lcd.write_string("_")` to clear the screen and always using ` lcd.clear()` prior to writing and never adding text to existing screen text. 

The system is now install. Next time it is down for maintenance I will check and try your suggestions.

Alan

alan-rpi, 2021-10-20:
I've discovered that if the last character of the text to be displayed is a blank then the screen briefly shows the text and then goes all white (as in the first post). 

In my case this is likely to happen where I programmatically generate the text and use spaces to pad out the text to get alignment. Luckily I push all my text to the screen via a common function so in that function I can test for a blank last character(s) and remove.