dbrgn/RPLCD #128: Could not display degree sign

daqingli616, 2021-09-06:
It seems that the latest version driver could not display the degree sign anymore. The following string can be printed out normally but could only display on LCD16x2 with degree sign missing.

uString = 'Temp: 30' + u'\N{DEGREE SIGN}' + 'C"
print (uString) shows: Temp: 30°C

lcd.write_string(uString) only display: Temp: 30 C

The test was done on a Raspberry 4 inside Python 3.7.3

dbrgn, 2021-09-06:
If you write "anymore", does that mean that it worked on a previous version? If yes, which version works and which version does not?

daqingli616, 2021-09-06:
Hi,

No, I did not get it to work. I saw a youtube video to demo its working but
it was published in 2016 and with the uchar(233) method. But that's with
Python2 and as Python3 changed uchar() to char() and I tried to use
char(233) but could not even print a degree sign.  I found out how to print
out a degree sign from online search with this u'\N{DEGREE SIGN}' method
but now I could print the degree sign inside a string and still the LCD
could not display it.

Thanks,

dbrgn, 2021-09-06:
That depends on the character map of your LCD. Did you see https://rplcd.readthedocs.io/en/latest/usage.html#character-maps?

daqingli616, 2021-09-06:
Hi,

After I add the charmap = 'A00' in my code, the degree sign shows up.

Thanks,

Daqing

closed 2024-04-04 by dbrgn