dbrgn/RPLCD #130: Issue on pi1

dreamonhigh, 2021-12-29:
hello, I love it. 

one thing I cannot solve. 
`#!/usr/bin/python3
from RPLCD.i2c import CharLCD
lcd = CharLCD('PCF8574', 0x27)

lcd = CharLCD(i2c_expander='PCF8574', address=0x27, port=0, \
              cols=20, rows=2, dotsize=8,charmap='A02', \
              auto_linebreaks=True, backlight_enabled=True)
lcd.clear()
lcd.cursor_pos = (0, 0)
lcd.write_string('hello ..')`

this script works fine on a **pi3** by setting "port=1". On **pi1** I have to set port=0 but this don't work. It shows self.SMBus(self._port) → FileNotFound: [Errno2] No such file or directory

PI1: i2cdetect -y 0 → shows 27 as I2C address 
PI3 i2cdecect -y 1 → shows 27 as I2C address 
but RPLCD.i2c/SMBus don't handle port=0 on PI1 the right way I think.
I also tried port=1 on PI1 with same error. by the way, setting port=0 on PI3 it shows same error. but using port=1 on PI3 works fine!

maybe a PI1 issue?

dreamonhigh, 2021-12-29:
I tried following example on PI1 and it worked with port=0 
So it seams to be a RPLCD issue. maybe default port=1 is always set?

`#!/usr/bin/python3

import smbus

bus = smbus.SMBus(0)    # 0 = /dev/i2c-0 (port I2C0), 1 = /dev/i2c-1 (port I2C1)

DEVICE_ADDRESS = 0x27      #7 bit address (will be left shifted to add the read >
DEVICE_REG_MODE1 = 0x00
DEVICE_REG_LEDOUT0 = 0x1d

#Write a single register
bus.write_byte_data(DEVICE_ADDRESS, DEVICE_REG_MODE1, 0x80)

#Write an array of registers
ledout_values = [0xff, 0xff, 0xff, 0xff, 0xff, 0xff]
bus.write_i2c_block_data(DEVICE_ADDRESS, DEVICE_REG_LEDOUT0, ledout_values)
`

dreamonhigh, 2021-12-29:
But testsuite works fine. I'm totally confused.  Here port=0 works on PI1..

./rplcd-test.py i2c testsuite expander=PCF8574 addr=0x27 port=0 cols=20 rows=4 charmap=A02

dreamonhigh, 2021-12-29:
my fault.
did "lcd = CharLCD" twice. so stupid.

Sorry

dbrgn, 2021-12-29:
@dreamonhigh haha, no worries 🙂

closed 2021-12-29 by dbrgn